sunspec/pysunspec2 #13: Publish to PyPI

altendky, 2020-11-12:
This project should be published to PyPI so proper version constraints can be used when referencing it and we don't have to deal with the Git URL.

altendky, 2021-02-17:
https://pypi.org/project/pysunspec/
https://pypi.org/project/pysunspec2/

altendky, 2021-03-25:
SunSpec has decided that they do not want me to have any access to the PyPI projects.  I have very little interest in dealing with their focus on control over progress.  I will be giving them ownership and expect to be immediately and permanently removed from both projects.  We'll see if they bother to make the libraries easier to use with releases on PyPI.

bobfox, 2021-03-25:
SunSpec is simply trying to correct the issue raised multiple times of github releases making it to PyPI in a standard way. This has nothing to do with any specific person and is just SunSpec trying own its product and take responsibility to correct a longstanding problem.

altendky, 2021-03-26:
I should have just linked over to the other issue and mentioned there was a new comment there.  Save everyone time by not duplicating...

https://github.com/sunspec/pysunspec/issues/41#issuecomment-807828623

closed 2021-04-27 by shelcrow