sunspec/pysunspec2 #22: Issue Scanning Device over TCP implementing Model 803

Paul-Paquette, 2021-02-03:
I'm trying to communicate with a Nuvation Low-Voltage BMS using this library over TCP. 
The device has the common model, and models 801, 802, and 803. I'm trying to reach information in model 803.
Inside Model 803 is a repeating block, which represents battery stacks.

Using the 'master' branch, I encountered issue #14 , so I figured I would switch to the development branch where this appears to be fixed.
When using the current 'development' branch, I encountered the following while scanning the device:
```
d.scan()
Traceback (most recent call last):

  File "<ipython-input-6-e84d88ff024c>", line 1, in <module>
    d.scan()

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\modbus\client.py", line 237, in scan
    model_id = mb.data_to_u16(model_id_data)

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\mb.py", line 112, in data_to_u16
    u16 = struct.unpack('>H', data[:2])

TypeError: a bytes-like object is required, not 'str'
```
After this error, I can see the library has recognized the device has the common model, models 801 and 802, but not 803. 
I can query what points are accessible, query/print values, etc. Have not tried writing any values - would like to confirm first I can read all information available on the device first. 

Could you advise next steps on this? Any advice on troubleshooting this or potential workarounds would be helpful.

bobfox, 2021-02-03:
What version of Python are you using?

Paul-Paquette, 2021-02-03:
I'm currently using Python 3.8.5. Would rolling back to an earlier Python 3 version (e.g. v3.6) be a good next step?

bobfox, 2021-02-03:
No, I don't think so, just wanted to make sure you were using Python 3.  We will take a closer look as this is a very new version.

bobfox, 2021-02-04:
In looking at the issue, we found what seems to be another issue that would not cause the behavior. But to be sure, we have updated the development branch. Can you retest and see if you still see the issue?

Paul-Paquette, 2021-02-04:
Trying this again today, I've found two different errors.

Initially with a scan (following a connect() statement), I get the following error:
```
d.scan()
Traceback (most recent call last):

  File "<ipython-input-6-e84d88ff024c>", line 1, in <module>
    d.scan()

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\modbus\client.py", line 255, in scan
    model = self.model_class(model_id=model_id, model_addr=addr, model_len=model_len, data=model_data,

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\modbus\client.py", line 160, in __init__
    SunSpecModbusClientGroup.__init__(self, gdef=gdef, model=self.model, model_offset=0, group_len=self.model_len,

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\modbus\client.py", line 69, in __init__
    device.Group.__init__(self, gdef=gdef, model=model, model_offset=model_offset, group_len=group_len,

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\device.py", line 326, in __init__
    g = self._init_repeating_group(gdef=gdef, model_offset=model_offset, data=gdata,

  File "C:\Users\paul\Anaconda3\lib\site-packages\sunspec2\device.py", line 487, in _init_repeating_group
    raise ModelError('Repeating group count not consistent with model length for model %s,'

ModelError: Repeating group count not consistent with model length for model 803,model repeating len = 6, model repeating group len = 32
```

If I call scan() again, I get the same error I saw yesterday. Which makes me suspect whether I had a connection issue when walking through to get a traceback, vs. when I was using my test script.

bobfox, 2021-02-04:
We have some some testing with the model configuration you reference but each implementation is different. Is it possible to get a register dump of the Modbus map in some form?

Paul-Paquette, 2021-02-04:
Sure. I polled all of the modbus registers claimed in their specification using modpoll. Attached are the register dump results, separated by model.
[nuvation_lv_bms_model_802.txt](https://github.com/sunspec/pysunspec2/files/5927488/nuvation_lv_bms_model_802.txt)
[nuvation_lv_bms_model_803.txt](https://github.com/sunspec/pysunspec2/files/5927489/nuvation_lv_bms_model_803.txt)
[nuvation_lv_bms_model_common.txt](https://github.com/sunspec/pysunspec2/files/5927490/nuvation_lv_bms_model_common.txt)
[nuvation_lv_bms_model_801.txt](https://github.com/sunspec/pysunspec2/files/5927491/nuvation_lv_bms_model_801.txt)

bobfox, 2021-02-05:
Looking the lengths of the models in the dumps you provided, they don't match the current content specified in the model definitions for 801, 802, and 803. Do you know if this is a current Nuvation implementation or is it possible it is an older development implementation that is using versions of the models before they were finalized?

Paul-Paquette, 2021-02-05:
Per their website, and consistent with their documentation, they state their devices are compliant with MESA Draft 3. I'm working to familiarize myself with these communication frameworks, but I believe this is several years old and has been superseded. 

I will follow up with the vendor to check if a product update exists to use a more recent model implementation.

In lieu of an update to the device, is it possible for me to choose what model definitions to use? Say I wanted to use the draft 3 versions of the models - is this possible given an appropriate model definition?

bobfox, 2021-02-05:
Only the current versions of the models are available in the models repository. That being said, all the changes should be reflected in GitHub. Unwinding it does not seem to be going in the right direction but is probably possible. This is an unfortunate situation due to deployment during model development but was commonly done with these models to enhance development. I think finding out if there is an update is a good first step.

Paul-Paquette, 2021-02-05:
Of course - the ideal outcome would be to update the device to use latest models. =)

What I'd like to get a sense of: Is it (currently) possible to use a _particular_ model definition at runtime, without rebuilding the package?
Say I had the definition file for Draft 3 of Model 803 in some location. 
Would there be any way to specify to use this _instead_ of the default definition for Model 803 (which is built into the package itself)?

bobfox, 2021-02-05:
Yes. device.py has the model_defs_path list that contains the directories and order to search for model definitions. Place the additional directories in this path list in the order that is appropriate. It searches them in the list order.

Paul-Paquette, 2021-02-05:
Thanks, Bob. I went back and found models in the models repo from 2015/2016 (44ccb15) (before the overhaul occurred for Draft 4). With those, and your pointer to where to edit the model search paths, I am able to read from this device without issue. 
I appreciate your help!

closed 2021-02-05 by Paul-Paquette