sunspec/pysunspec2 #28: Easier usage would be welcome.

smith9, 2021-03-21:
Hi,

I like that you have added the pysunspec to PYPi and this comment isn't going to come across well, so please don't take it to badly.
From an outsiders perspective
1) The doco says this project has some differences from v1 and because of that, it is not backwards compatible.
From my point of view, it looks like a total re-write by a different team of developers and has breaking changes at all level of the code and that If I want to output the data from my inverter, my reading code will be a complete rewrite.

2) I had a simple read code from v1 that was loops within loops. The model was generic and so to was converting the data from one format to another. The whole model -> Block -> Points was easy to visualise in ones head. Now the blocks are called groups but they are now dictionaries, with a group called "Curve", which isn't too intuitive

3) When installing from PYPi, the models still have to be downloaded from github and placed manually in sitepackages. - this makes it a real pain to install.

4) Before you had a nice example on one page of looping over the Model -> Block ->Points, but don't see any nice examples in v2. I was looking at the to_spreadsheet to get some inspiration, but it just looks overly complex. My brain is having a much harder time following the V2 code compared to the v1 code.

5) It looks like there was concentration on accessing the data like d.DERMeasureAC[0].LLV.value which is nice, but I am trying to dump all the values from my inverter and store the output in simpler json files and using marshmallow to read those for processing like uploading to PVOutput. So accessing in that way is not really helpful.

Do you have an code examples like before that give a way to iterate over all the data and dump it out?

bobfox, 2021-03-22:
Are you trying to support SunSpec information models generically or are there specific models you are interested in?

Here is an example of getting model 701 contents in json. If you want computed values with the scale factor applied, you can set computed=True as in this example. If you want the base values, remove that.

import sunspec2.modbus.client as client
d = client.SunSpecModbusClientDeviceTCP(slave_id=1, ipaddr='127.0.0.1', ipport=8502)
d.scan()
model = d.models[701][0]
print(model.get_json(computed=True))

smith9, 2021-03-23:
Hi,

Yes, that is super helpful, was able to use that against a device

device.get_json(computed=True)

Will be much easier for me to write code against a simple json structure
than an nested object model.

What I am doing is extracting all the info from my ABB Inverter and dumping
it to a json file.
Then uploading to pvoutput from the data in the file
(if internet is not available, then a bunch of files will build up and once
internet is available again - will do a bulk upload to pvoutput)

It is also interesting to keep all the files to see if anything in the
inverter is degrading over time - but never got around to coding that part
so far.

I can see in the codebase a test file for device_1547.json which looks like
the exact format that the device.get_json outputs - so that will make it
easy for me to generate test data for running against that came from a real
inverter.

Would be good for the get_json to be added to the readme file of pysunspec2.
If you are looking for other suggestions, then something like the below
would also be helpful to users
m = model("model101.json")
units = m.getUnits("WH")
descr = m.getDesc("WH")

Thanks
David

closed 2021-10-13 by shelcrow