sunspec/pysunspec2 #36: Modbus exception with d.scan() on SMA inverter

bijwaard, 2021-05-26:
Dear developers,
I get a Modbus exception with v1.0.4 (both via pip3 install as well as from master branch) when trying to scan() an SMA inverter (STP-5000TL-20 via Ubuntu 20.04.02 LTS), see below:
$ python3
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sunspec2.modbus.client as client
>>> d = client.SunSpecModbusClientDeviceTCP(slave_id=126, ipaddr='192.168.0.125', ipport=502)
>>> d.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dennis/.local/lib/python3.8/site-packages/sunspec2/modbus/client.py", line 262, in scan
    model.read()
  File "/home/dennis/.local/lib/python3.8/site-packages/sunspec2/modbus/client.py", line 85, in read
    data = self.model.device.read(self.model.model_addr + self.offset, self.len)
  File "/home/dennis/.local/lib/python3.8/site-packages/sunspec2/modbus/client.py", line 317, in read
    return self.client.read(addr, count, op)
  File "/home/dennis/.local/lib/python3.8/site-packages/sunspec2/modbus/modbus.py", line 584, in read
    data = self._read(addr + read_offset, read_count, op=op)
  File "/home/dennis/.local/lib/python3.8/site-packages/sunspec2/modbus/modbus.py", line 545, in _read
    raise ModbusClientException('Modbus exception %d: addr: %s count: %s' % (except_code, addr, count))
sunspec2.modbus.modbus.ModbusClientException: Modbus exception 2: addr: 40085 count: 98

Any idea how to solve this?

Kind regards,
Dennis

bijwaard, 2021-05-26:
Hi,
I figured out to the use tracefunc, and get the following result when I enable it with a simple print function:

`
d = client.SunSpecModbusClientDeviceTCP(slave_id=126, ipaddr='192.168.0.125', ipport=502,trace_func=print)
`

When only reading the first 11 bytes from register 40085 everything goes well:

> d.read(40085,11)
> 0000000000067E039C95000B
< 0000000000197E0316000C0062000000000000000000010000000500010000
b'\x00\x0c\x00b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x05\x00\x01\x00\x00'

When reading one more byte, I trigger the issue:

> d.read(40085,12)
> 0000000000067E039C95000C
< 0000000000037E8302
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/client.py", line 317, in read
    return self.client.read(addr, count, op)
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/modbus.py", line 584, in read
    data = self._read(addr + read_offset, read_count, op=op)
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/modbus.py", line 545, in _read
    raise ModbusClientException('Modbus exception %d: addr: %s count: %s, len(resp)=%d, resp[TCP_HDR_LEN+1]=0x%x' % (except_code, addr, count, len(resp), resp[TCP_HDR_LEN + 1]))
sunspec2.modbus.modbus.ModbusClientException: Modbus exception 2: addr: 40085 count: 12, len(resp)=9, resp[TCP_HDR_LEN+1]=0x83

When doing the complete scan, I get:

> d.scan()
> 0000000000067E0300000003
< 0000000000097E0306FFFFFFFFFFFF
> 0000000000067E039C400003
< 0000000000097E030653756E530001
> 0000000000067E039C430001
< 0000000000057E03020042
> 0000000000067E039C420042
< 0000000000877E038400010042534D410000000000000000000000000000000000000000000000000000000000536F6C617220496E76657274657200000000000000000000000000000000000039303938000000000000000000000000333931393133303000000000000000003330353133313932320000000000000000000000000000000000000000000000
> 0000000000067E039C860001
< 0000000000057E0302000B
> 0000000000067E039C870001
< 0000000000057E0302000D
> 0000000000067E039C86000D
< 00000000001D7E031A000B000D00640003000100000040AD9FD6920000000000000000
> 0000000000067E039C950001
< 0000000000057E0302000C
> 0000000000067E039C960001
< 0000000000057E03020062
> 0000000000067E039C950062
< 0000000000037E8302
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/client.py", line 262, in scan
    model.read()
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/client.py", line 85, in read
    data = self.model.device.read(self.model.model_addr + self.offset, self.len)
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/client.py", line 317, in read
    return self.client.read(addr, count, op)
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/modbus.py", line 584, in read
    data = self._read(addr + read_offset, read_count, op=op)
  File "/home/dennis/compiling/SST/pysunspec2/sunspec2/modbus/modbus.py", line 545, in _read
    raise ModbusClientException('Modbus exception %d: addr: %s count: %s, len(resp)=%d, resp[TCP_HDR_LEN+1]=0x%x' % (except_code, addr, count, len(resp), resp[TCP_HDR_LEN + 1]))
sunspec2.modbus.modbus.ModbusClientException: Modbus exception 2: addr: 40085 count: 98, len(resp)=9, resp[TCP_HDR_LEN+1]=0x83

jayatsandia, 2021-05-26:
This SMA device is likely not pySunSpec2-compliant.  You'll have to use the original pysunspec python library because it is not backward compatible.  There were major revisions to the structure of the datatypes when SunSpec moved to pysunspec2.  Try: https://github.com/sunspec/pysunspec

bobfox, 2021-05-26:
We will look at this in more detail but wanted to clarify the comment from @jayatsandia. pysunspec2 should be compatible with all SunSpec compliant devices and is backward compatible with all SunSpec standard model definitions and implementations. We recommend that you replace pysunspec with pysunspec2 for SunSpec device interaction. Support for pysunpec is very limited. The scripting syntax when interacting with devices is a little different in pysunspec2 than pysunspec so minor adjustments may be needed to scripts moving from pysunspec to pysunspec2.

SunSpec has updated the modeling and model definitions but still supports all existing model definitions.

jayatsandia, 2021-05-26:
My mistake.  I misunderstood the following statement in the pySunSpec2 README: "The Python objects used for interacting with devices have some differences from version 1 and it is not backward compatible with version 1. This version of pySunSpec must be used with devices implementing the 7xx series of information models as they use updated modeling concepts."

bobfox, 2021-05-26:
No worries, good to clarify.

bijwaard, 2021-05-27:
Hi again,
I created a simple Python program [sma-test.py.txt](https://github.com/sunspec/pysunspec2/files/6554657/sma-test.py.txt) to test the addresses and length fields for the different modules and this appears to work fine for my SMA inverter:
- checkFromPICS(): start registers from each module sheet of the PICS file
- checkWithOffsets(): using length registers when starting with register 40002 for the first ID, and length on the following register address. It stops when length=0

Here is the output for the offset test:

$ ./sma-test.py 
40002: ID=1
40003: L=66
40070: ID=11
40071: L=13
40085: ID=12
40086: L=98
40185: ID=103
40186: L=50
40237: ID=120
40238: L=26
40265: ID=121
40266: L=30
40297: ID=122
40298: L=44
40343: ID=123
40344: L=24
40369: ID=124
40370: L=24
40395: ID=126
40396: L=64
40461: ID=127
40462: L=10
40473: ID=128
40474: L=14
40489: ID=131
40490: L=64
40555: ID=132
40556: L=64
40621: ID=160
40622: L=48
40671: ID=4335
40672: L=0

Kind regards,
Dennis

Epyon01P, 2021-09-27:
Can confirm this issue also happens on SMA SB 4000 TL, SB 1500, SB2000 TL and SB Storage 2500. So probably all SMA devices. Communication works just fine when using the old pysunspec library.

moorboss, 2021-10-12:
> Can confirm this issue also happens on SMA SB 4000 TL, SB 1500, SB2000 TL and SB Storage 2500. So probably communication wih all SMA devices is affected. Communication works fine when using the old pysunspec library.

It looks like I have the same problem with my SMA SB 3000TL-21 
How can you use the old pysunspec library? Which file(s) do you need for that?

shelcrow, 2021-10-13:
@bijwaard There was an issue with models reads being two registers short that we fixed in the latest release. Can you update pysunspec2 and try again?