sunspec/pysunspec2 #38: Modbus client does not comply with the standard limit of registers for read/write requests

jmmgenerac, 2021-05-30:
The [Modbus specification](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf) states that the maximum number of registers returned from a Function 3 ("Read Holding Registers") or Function 4 ("Read Input Registers") is 125.
And the number of registers for a Function 16 ("Write Multiple registers") is 123.

The pysunspec2 Modbus client only supports one value for the max register count. By default it is 125. Because there is only one value, this means that there is no way for the client to reach both limits of the standard at the same time.

Could the client be modified to comply with the standard? I think either of the following would be fine:
* The default max number of registers for a write is by 123 registers, and default max for a read is 125
* When instantiating the client, the max count for write can be configured separately from max count for read

bobfox, 2021-06-01:
We will add REQ_WRITE_COUNT_MAX = 123 and adjust the write functions to use this a the default value.

I am not sure I understand the second bullet item, are you referring to a configuration point in pysunspec2?

jmmgenerac, 2021-06-01:
Thanks, @bobfox. That will cover the need perfectly.