sunspec/pysunspec2 #41: Is there a limit on number of points in a group?

jmmgenerac, 2021-06-14:
I have a vendor model that was converted from SMDX. It has a repeating block that contains 128 points of type uint16. When I scan the device that exposes this model (which, FWIW, loads just fine in pysunspec), I get the following error:
```
>>> c.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/sunspec2/modbus/client.py", line 260, in scan
    model = self.model_class(model_id=model_id, model_addr=addr, model_len=model_len, data=model_data,
  File "/usr/lib/python3.8/site-packages/sunspec2/modbus/client.py", line 160, in __init__
    SunSpecModbusClientGroup.__init__(self, gdef=gdef, model=self.model, model_offset=0, group_len=self.model_len,
  File "/usr/lib/python3.8/site-packages/sunspec2/modbus/client.py", line 69, in __init__
    device.Group.__init__(self, gdef=gdef, model=model, model_offset=model_offset, group_len=group_len,
  File "/usr/lib/python3.8/site-packages/sunspec2/device.py", line 401, in __init__
    raise ModelError('Nested groups too big')
sunspec2.device.ModelError: Nested groups too big
```

This is apparently due to a comparison of the  "group length" with the literal `ACCESS_REGION_REGS`, which has a value of 123 (remember, there are 128 points in my model's repeating block).

Can you explain the reason for this 123-register limitation? Would it be feasible to change the library to allow models having groups that contain more registers than this?

bobfox, 2021-06-15:
As a simplification, groups are used to divide the models that require multiple accesses to read or write the complete model.  The length of 123 registers is used as it is the shorter length between Modbus max read and max write.

We will take a look at the logic and see what assumptions are being made about group access in a single operation.

closed 2021-12-10 by shelcrow

bobfox, 2022-01-08:
Added addition correction to initial update to handle non-repeating models correctly.