sunspec/pysunspec2 #86: SetuptoolsDeprecationWarning: setup.py install is deprecated.

tk1, 2024-01-02:
When trying to install with 

    python .\setup.py install

I get the warning:  SetuptoolsDeprecationWarning: setup.py install is deprecated.

ttansy, 2024-01-02:
pysunspec needs maintenance.

Tom Tansy
SunSpec Alliance
831-227-1073
***@***.***

ttansy, 2024-01-03:
Hi Tom,

It seems like the &#39;python setup.py install&#39; has been deprecated. Here&#39;s the article talking about it: https://packaging.python.org/en/latest/discussions/setup-py-deprecated/

I believe we just need to change the Readme on github to state &#39;Use pip install pysunspec2&#39; for installation. That&#39;s the recommended command for installation and it works fine. I will make that change.

Thanks,
Kudrat

Kudrat Kaur
***@***.***

SunSpec Alliance

Desk tracking: [DESK-3b2n-2nqmh]

ttansy, 2024-01-03:
Sounds good Kudrat. Thanks for making the change.

Let's make a plan for supporting this project. Should we co-opt the Modbus
group once per month?

Tom Tansy
SunSpec Alliance
831-227-1073
***@***.***

tk1, 2024-01-09:
> Hi Tom, It seems like the &#39;python setup.py install&#39; has been deprecated. 

I installed it via pip, but my intention was to say that the documentation is outdated.

Kudrat9, 2024-01-09:
Hi @tk1 Thanks for pointing that out. We've updated the Readme to install using pip and mentioned the python setup.py install is deprecated.

closed 2024-01-09 by Kudrat9